HandHeldLegend/HOJA-LIB-ESP32 #8: Second esp32

IgnacioGV, 2022-11-02:
Hi, first of all your project is awesome and in trying to reproduce the NES controller but I had an inconvenient with a second esp32, after programming it mentioned that the authentication with the switch was successful and appear a code but when I try to find it in the switch this one never appears. do you know what is happening or only I can use one esp32 per Nintendo switch?

mitchellcairns, 2022-11-02:
I'll have to look into this more in the coming weeks. The Mac address on a new flash would be randomized on the first boot, so it shouldn't be a conflict there. Maybe there's something else going on, but I'll have to look into this further later on. You might try monitoring the controller while connecting or attempting and seeing what happens. Thank you.

mitchellcairns, 2022-11-30:
> Hi, first of all your project is awesome and in trying to reproduce the NES controller but I had an inconvenient with a second esp32, after programming it mentioned that the authentication with the [...]

Hi Ignacio. I've tested with the latest version of code, verified two ESP32 does work.

closed 2022-11-30 by mitchellcairns