HandHeldLegend/HOJA-LIB-ESP32 #12: ESP32-based controller converter?

TheChatty, 2023-01-12:
This project is already able to emulate a Switch Pro Controller. Is it possible to have different brand controllers paired, convert their protocol and feed it to the emulator like the 8BitDo USB dongle does? The emulator would then be paired to a real Switch console.

mitchellcairns, 2023-01-12:
It's certainly possible but it's out of the scope and unplanned for this project. Let me know if you have any other questions on this!

TheChatty, 2023-01-13:
Since documentation is lacking at the moment: how do you use your current piece of code?

mitchellcairns, 2023-01-13:
> Since documentation is lacking at the moment: how do you use your current piece of code?

Documentation is being worked on, nothing public yet please stay tuned should be something in the next two weeks.

mitchellcairns, 2023-01-15:
> Since documentation is lacking at the moment: how do you use your current piece of code?

I have created one example project with some detailed commentary for learning purposes. Let me know if this helps
https://github.com/HandHeldLegend/HOJA-LIB-ESP32/tree/main/examples/basic-wired-gamepad

TheChatty, 2023-01-15:
I'm missing an end-to-end example.

* Does this project enable oneself to push push a button (shorten the defined GPIOs) and one can see the result on one's Switch console?
* If yes, what steps a necessary to accomplish this (mainly how does the pairing work)?
* Are there any nightlies I can flash my ESP32 to avoid the hassle of setting up a working IDE?
* Please add [gitpod.io](https://gitpod.io/#https://github.com/HandHeldLegend/HOJA-LIB-ESP32/) compatibilty for easy compilation

mitchellcairns, 2023-01-15:
> I'm missing an end-to-end example.
>
> * Does this project enable oneself to push push a button (shorten the defined GPIOs) and one can see the result on one's Switch console?
> * If yes, what steps a [...]

Until I have the time to get all of the full documentation together, I recommend looking through the code using intellisense in VS Code IDE. I can certainly look into gitpod.io

No nightlies at the moment. Please take a close look at the example code, read all of the notes and let me know if you have any questions.

TheChatty, 2023-01-24:
I've compiled a few ESP8266/ESP32 projects already (Tasmota, OpenMQTTGateway, ...) mostly using gitpod.io but I'm unable to get this compiled. And I'm still wondering how a useful end-to-end scenario would look like. Maybe @JPZV could shed a little light on how he's using it?

mitchellcairns, 2023-01-24:
> I've compiled a few ESP8266/ESP32 projects already (Tasmota, OpenMQTTGateway, ...) mostly using gitpod.io but I'm unable to get this compiled. And I'm still wondering how a useful end-to-end scenario would look like. Maybe @JPZV could shed a little light on how he's using it?

What build errors are you getting?

Thanks

TheChatty, 2023-01-25:
Well... I using my above link to gitpod which creates a remote IDE and this source code. I then tried to manually install all required dependencies in the remote shell but eventually failed to setup ESP-IDF correctly (openocd/libusb issue).

mitchellcairns, 2023-01-25:
It looks like you're trying to build the component-- try running the build command from the example directory and let me know if that works

mitchellcairns, 2023-01-25:
> That was it and now it worked and I got the `basic-wired-gamepad.bin`. This proves compilation with gitpod is easily possible (requires IDF5 and libusb installation). Now if this project would be set up correctly then all that would be required is to follow my above link, wait for automatic setup to finish and download and flash the compiled bin.
> 
> Still needed:
> 
> * [Gitpodifying](https://www.gitpod.io/guides/gitpodify)
> * reasonable [.gitpod.Dockerfiler](https://github.com/ferrous-systems/espressif-trainings/blob/main/.gitpod.Dockerfile) template

It's something I will look into when I have more time available, I appreciate the effort and the recommendation!

TheChatty, 2023-01-25:
Isn't the bootloader part of the IDF? In any case it's also present in the build directory.